LongSoft/UEFITool #431: Do you wish to fix it in nvramparser.cpp?

Ykidia, 2025-05-05:
nvramparser.spp @ 992:

```
            if (!outerPadding.isEmpty()) {
                info = usprintf("Full size: %Xh (%u)", (UINT32)outerPadding.size(), (UINT32)outerPadding.size());
                model->addItem(localOffset + previousStoreEndOffset, Types::Padding, getPaddingType(outerPadding), UString("Padding"), UString(), info, UByteArray(), outerPadding, UByteArray(), Fixed, index);
                outerPadding.clear();
            }
            
            // Construct header and body
            header = volumeBody.left(storeOffset + sizeof(PHOENIX_FLASH_MAP_HEADER));
            body = volumeBody.mid(storeOffset + header.size(), storeSize - header.size());

            // Add info
            name = UString("Phoenix SCT flash map");
            info = usprintf("Signature: _FLASH_MAP\nFull size: %Xh (%u)\nHeader size: %Xh (%u)\nBody size: %Xh (%u)\nEntries: %u\nReserved: %08Xh",
                                    storeSize, storeSize,
                                    (UINT32)header.size(), (UINT32)header.size(),
                                    (UINT32)body.size(), (UINT32)body.size(),
                                    parsed.num_entries(),
                                    parsed.reserved());

```

I think that header and body contents are incorrect. Please consider this fix:

```
            }
            
            // Construct header and body
-            header = volumeBody.left(storeOffset + sizeof(PHOENIX_FLASH_MAP_HEADER));
+            header = volumeBody.mid(storeOffset, sizeof(PHOENIX_FLASH_MAP_HEADER));
            body = volumeBody.mid(storeOffset + header.size(), storeSize - header.size());

            // Add info
            name = UString("Phoenix SCT flash map");

```

If you think that this is not an error, then count: body will have invalid (offset with TWICE storeOffset) and therefore invalid size which will result in always overflowing storeSize: storeSize < storeOffset + header.size() + header.body().

NikolajSchlej, 2025-05-05:
Sure, it's a copy-paste error, will be fixed.

NikolajSchlej, 2025-05-05:
Fixed by https://github.com/LongSoft/UEFITool/commit/c0368902c4cc7982eae767768f70b49e580b4bca

closed 2025-05-05 by NikolajSchlej