LongSoft/UEFITool #435: Can't replace individual entries in "EfiSystemNvDataFvGuid Volume NVRAM"

schildbach, 2025-06-09:
"UEFITool NE" shows the contents of this section just fine.

uefitool 1.28 with the old engine only shows "unknown" and can't drill further into the volume. The prevents me from replacing invidual entries in the volume.

Is there any alternative tool that can do this?

closed 2025-06-10 by NikolajSchlej

schildbach, 2025-06-10:
> Try [UefiBiosEditor](https://drive.google.com/file/d/1Ib9B_4Chnm77_8BCbcCuTxaykEVyFYT7/view).

Hu, is this a malware link? A naked .exe? Is there source code? A project page? How is this related to UEFITool?

gregsskyles, 2025-06-10:
FYI, the link just leads back here: `https://github.com/LongSoft/UEFITool/issues/html`

NikolajSchlej, 2025-06-10:
Fixed, the correct link is http://iorlvskyo.blogspot.com/2016/06/uefi-bios-editor.html