LongSoft/UEFITool #439: Should we switch to docked widgets instead of fixed UI areas?

NikolajSchlej, 2025-06-30:
@Ykidia suggested in his recent PR to switch from the current UI to a docked UI, examples can be found in https://github.com/LongSoft/UEFITool/pull/438. I've merged it initially for testing, but found out that the docks look somewhat ugly and foreign in macOS, and the app keeps throwing EXC_BAD_ACCESS during exiting.

Due to not being fully onboard with the PR as it stands, I've reverted it, and took only the non-UI additions and fixes.

I'l like to use this issue for further discussion on that idea. CC @vit9696

vit9696, 2025-07-01:
I believe the feature @Ykidia suggested may be fine, but let's start with having something working first.

Ykidia, 2025-07-05:
Fixed that. Looks like a bug in Qt. Also reduced ugly look by hiding splitters, because now splitters can be more or less than two as in classic UI, and they everywhere between docks.
Added deleting search and goto dialogs - hello, memory leak!
And, by the way, fixed "recusrively" to "recursively" :)
Would you like to reopen PR or wait while all ugliness will be fixed? I may need much more time for this, but the UI will still not be the same for objective reasons (docks looks different from fixed elements by design).

> I believe the feature [@Ykidia](https://github.com/Ykidia) suggested may be fine, but let's start with having something working first.

It would be nice to see and feel it in action to have your own solid opinion, isn't it? As they say, **try before buy** ;) (or refuse)

P.S. Qt maintainers themselves prefer to use ADS instead of their own QDockWidgets, but the latter still have a right to exist and to be used.
P.P.S. Qt Creator using ADS looks some ugly too on macOS despite the coolness of this library. I was a little disappointed by this - couldn't they have tried harder? :)

NikolajSchlej, 2025-07-06:
Please open a new PR with the updated code against the current ToT.

NikolajSchlej, 2025-07-09:
#440 should be a fix for this, let's test it once again and see if this can be closed.

NikolajSchlej, 2025-07-31:
Tested, can be closed as completed. One more thing needs to be done before releasing this as part of A73 - a new "View->Reset widget layout to default" menu item.

closed 2025-07-31 by NikolajSchlej